BiancoRoyal/node-red-contrib-modbus #416: after modbus failure no initialize possible

Becker884, 2023-10-08:
### Which node-red-contrib-modbus version are you using?

5.26.0

### What happened?

Hi, I used for years a RS485 <-> USB dongle (DSD TECH SH-U11F) to query 3x Eastron SDM72v2 on a RPi4 with Node-Red.
Sometimes I got a modbus failure, but it doesn´t matter, after 5s modbus reconnected and it runs.

I replaced the RPi4 with a RPi2 - same flow, same OS, same Node-Red Version.

Sometimes I get also a modbus failure.
`Oct 8 20:22:15 raspberrypi2 Node-RED[1605]: 8 Oct 20:22:15 - [warn] [modbus-getter:0006 - 0035] Modbus Failure On State sending Get More About It By Logging

Oct 8 20:22:15 raspberrypi2 Node-RED[1605]: 8 Oct 20:22:15 - [error] [modbus-getter:0060 - 0063] Error: Timed out`

but after that it doesn´t reconnect / initialize, I have to stop/start Node-Red. I tried other usb ports, different settings (9600B / 19200B - queue delay 10ms or 50ms). I switched the DSD Tech with an other one.

Any ideas why it doesn´t initialize ?
[image]

### Server

None/This is related to a node that doesn't connect to a server

### How can this be reproduced?

can´t be reproduced, the failure comes sometimes, at least 12h.

### What did you expect to happen?

re-initialize

boogiebug, 2023-10-22:
I got the same situaltion but with different dongle model. I've successfully reinitialized the port by running `usbrest` to the target port from command line.

Becker884, 2023-10-22:
I tried another RPi 3 with NodeJS 18 - no problems.

the RPi 2 was an old one with Cortex-A7 - I don't know if that was the reason.

NodeJS 20 is not released, and with NodeJS 20 the Modbus RTU doesn´t work.

biancode, 2023-10-23:
sometimes it can be the serialport package itself see https://serialport.io/docs/11.x.x/guide-platform-support

closed 2024-01-22 by github-actions[bot]