BiancoRoyal/node-red-contrib-modbus #420: Modbus Flex Sequencer broken in 5.26?

hoegge, 2023-10-28:
### Which node-red-contrib-modbus version are you using?

5.26 - newest in node-red pallete

### What happened?

After upgrading from 5.23 all my Modbus flows has stopped. Nothing gets through the Mobus Flex Sequencer It simply does nothing. The Modbus Flex Getter works fine.

It seems after a lot of troubleshooting, that the Flex Sequencer only fires if `msg.payload` contains something? Otherwise it ignores input.

### Server

Other/External server

### How can this be reproduced?

Use in in Node-red 3.0.2

hoegge, 2023-10-28:
It seems to have happened here: https://github.com/BiancoRoyal/node-red-contrib-modbus/commit/b7acd97ab8f940decb0889f9bf8125fa278f1f57
where this was introduced here: https://github.com/BiancoRoyal/node-red-contrib-modbus/commit/758a3a595b4ff9bb2980f3f0bd91aa1d911f8177#r131145217

Seems to be a breaking change, since documentation does not require an `msg.payload` input but `msg.messages`

grewek, 2023-12-05:
Hi @hoegge 

Thank you for the report, i have looked at the first issues and it seems to me that this is intended behaviour as we do not want the node to work on an empty payload.

biancode, 2024-01-28:
You can inject a timestamp and not empty msg and it works in our testing as expected.

hoegge, 2024-01-28:
Ok / thanks for clarification @grewek and @biancode - But why do you want a `msg.payload` when the documentation only says you want a `msg.messages`. Then this should be added as a required item in the docs? It is a breaking change - so should at least be documented. I have tried to submit clarified documentation to this project before just to be rejected :-(

biancode, 2024-01-31:
@hoegge thank you for the question and we had not enough time invested until know to change that node to our defaults. At the moment the code says it still `msg.sequences` and payload is checked from a first change from us as all nodes do the check. Next step would to set sequences into msg.payload. I think that node needs a bit more refactoring work to stay compatible to existing flows.

closed 2024-08-18 by github-actions[bot]