BiancoRoyal/node-red-contrib-modbus #425: Write input register don‘t work

PeMi68, 2023-12-11:
### Which node-red-contrib-modbus version are you using?

5.25

### What happened?

If I try to write with function into input register, only write at adress 0 work. If I try to write to any other adress, no value is written.

### Server

Modbus-Server Node

### How can this be reproduced?

Try function with input from example.

biancode, 2024-01-27:
Hi, Input Register can be write protected by the server - I think Holding Registers are better in your case.

biancode, 2024-01-28:
I tested it and it works as expected and we have all basic features in e2e testings for every release by CI/CD.

closed 2024-04-28 by github-actions[bot]