BiancoRoyal/node-red-contrib-modbus #426: yarn install error because url have bad charactor

lrobot, 2023-12-13:
### Which node-red-contrib-modbus version are you using?

yes

### What happened?

```
$ yarn add node-red-contrib-modbus                                                    
➤ YN0000: ┌ Resolution step
➤ YN0001: │ Error: @open-p4nr/modbus-serial@https://dl.cloudsmith.io/public/iniationware-gmbh/plus4nodered/npm/open-p4nr/modbus-serial/8.1.0/modbus-serial-8.1.0.tgz#_gl=1*oezs8a*_gcl_au*NDk2NjE5MzkwLjE2ODc1MDQxODk. isn't supported by any available resolver
    at xf.getResolverByDescriptor (~/.cache/node/corepack/yarn/3.5.0/yarn.js:391:1647)
```

### Server

Modbus-Server Node

### How can this be reproduced?

yarn3 run : yarn add node-red-contrib-modbus       

### What did you expect to happen?

install sucess

### Other Information

but install failed

lrobot, 2023-12-13:
yarn add '@open-p4nr/modbus-serial@https://dl.cloudsmith.io/public/iniationware-gmbh/plus4nodered/npm/open-p4nr/modbus-serial/8.1.0/modbus-serial-8.1.0.tgz#_gl=1*oezs8a*_gcl_au*NDk2NjE5MzkwLjE2ODc1MDQxODk' also failed 
yarn add '@open-p4nr/modbus-serial@https://dl.cloudsmith.io/public/iniationware-gmbh/plus4nodered/npm/open-p4nr/modbus-serial/8.1.0/modbus-serial-8.1.0.tgz'.  can sucess

biancode, 2024-01-28:
please try V5.28.0 soon

closed 2024-04-28 by github-actions[bot]