BiancoRoyal/node-red-contrib-modbus #429: Function Code 43

innoagg, 2024-01-26:
### Which node-red-contrib-modbus version are you using?

Latest

### What happened?

Is there a cheat to be able to use function code 43 or others? Is it intentional to filter certain function codes? Flex Getter will only accept common ones. 

[image]

### Server

Modbus-Flex-Server Node (Please attach Script to the next section)

### How can this be reproduced?

Use a function code 43 in "payload.fc"

biancode, 2024-01-27:
Hi, you can try the CustomFC node to test with but it needs some function around to handle the buffers and encapsulation

innoagg, 2024-01-27:
I apologize for not seeing that node. I must have read right over it. I'll give it a shot.

closed 2024-01-27 by innoagg

biancode, 2024-01-27:
right the name is in that version Modbus Flex FC
[image]