BiancoRoyal/node-red-contrib-modbus #434: need 'NaN' string instead of 0 when no value written to modbus flex server

vnvasireddy, 2024-01-29:
### Describe the requested feature

Dear team,
As we are using node-red modbus package for production, we are integrating it with our controller where our controller communicates with modbus flex server. When there is no data written to modbus flex server we need string 'NaN' instead of '0'.
If 0 is reading controller take it as data and functioning is wrong. So can you add this feature or any suggestion to get NaN from server if no data.

Thanks in advance

### Motivation

immediate

biancode, 2024-01-31:
Hi @vnvasireddy, the suggestion here would be an exception to send if the value was not written. That is a more standard conform way of handling errors. An exception can tell what was wrong.

vnvasireddy, 2024-02-01:
Thank you for quick reply...
Any suggestion how to send an exception. I mean any example flow or any syntax to send

biancode, 2024-02-05:
At the moment I mean there is just the option to go ahead and to finish the server in the modbus-serial package to handle exceptions or to allow that.  `if ... else exception`

closed 2024-06-21 by github-actions[bot]