BiancoRoyal/node-red-contrib-modbus #436: Flex Getter does not reconnect to serial port after redeploy

serpent213, 2024-01-29:
### Which node-red-contrib-modbus version are you using?

5.28.0

### What happened?

Log during redeploy:

```
2024-01-29T08:12:11.405Z contribModbus:queue:core queue serial lock command node name: Modbus RS485 id: e39ff271e27b2afe
2024-01-29T08:12:11.415Z contribModbus:queue:core sequential de-queue command
2024-01-29T08:12:11.417Z modbus-serial {
  action: 'send serial rtu buffered',
  data: <Buffer 02 03 00 04 00 01 c5 f8>,
  unitid: 2,
  functionCode: 3,
  length: 7
}
2024-01-29T08:12:11.452Z modbus-serial {
  action: 'receive serial rtu buffered port',
  data: <Buffer 02 03 02 00 89 3d e2>,
  buffer: <Buffer 02 03 02 00 89 3d e2>
}
2024-01-29T08:12:11.452Z modbus-serial {
  action: 'emit data serial rtu buffered port',
  buffer: <Buffer 02 03 02 00 89 3d e2>
}
2024-01-29T08:12:11.453Z contribModbus:queue:core queue serial unlock command node name: Modbus RS485 id: e39ff271e27b2afe
2024-01-29T08:12:11.514Z contribModbus:queue:core queue serial lock command node name: Modbus RS485 id: e39ff271e27b2afe
2024-01-29T08:12:11.620Z contribModbus:queue:core sequential de-queue command
2024-01-29T08:12:11.623Z modbus-serial {
  action: 'send serial rtu buffered',
  data: <Buffer 01 0f 00 00 00 08 01 01 3f 55>,
  unitid: 1,
  functionCode: 15,
  length: 8
}
29 Jan 09:12:11 - [debug] saved flow revision: 2b39acde3730fc9777f701abe39d70f4
29 Jan 09:12:11 - [info] Stopping modified flows
29 Jan 09:12:11 - [debug] red/nodes/flows.stop : stopping flow : b7fd3cfd8418de99
29 Jan 09:12:11 - [debug] red/nodes/flows.stop : stopping flow : dc8b2f6a2286fcc4
29 Jan 09:12:11 - [debug] red/nodes/flows.stop : stopping flow : 1fa0281e7267a0a1
29 Jan 09:12:11 - [debug] red/nodes/flows.stop : stopping flow : global
29 Jan 09:12:11 - [info] Stopped modified flows
29 Jan 09:12:11 - [info] Updated flows
2024-01-29T08:12:11.655Z modbus-serial {
  action: 'receive serial rtu buffered port',
  data: <Buffer 01 0f 00 00 00 08>,
  buffer: <Buffer 01 0f 00 00 00 08>
}
29 Jan 09:12:11 - [info] Starting modified flows
29 Jan 09:12:11 - [debug] [inject:Zyklus] repeat = 2500
29 Jan 09:12:11 - [info] Started modified flows
2024-01-29T08:12:11.669Z modbus-serial {
  action: 'receive serial rtu buffered port',
  data: <Buffer 54 0d>,
  buffer: <Buffer 01 0f 00 00 00 08 54 0d>
}
2024-01-29T08:12:11.669Z modbus-serial {
  action: 'emit data serial rtu buffered port',
  buffer: <Buffer 01 0f 00 00 00 08 54 0d>
}
2024-01-29T08:12:11.669Z contribModbus:queue:core queue serial unlock command node name: Modbus RS485 id: e39ff271e27b2afe
2024-01-29T08:12:13.667Z modbus-serial {
  action: 'port open error',
  error: [Error: Error Resource temporarily unavailable Cannot lock port]
}
2024-01-29T08:12:13.667Z contribModbus:queue:core queue serial unlock command node name: Modbus RS485 id: e39ff271e27b2afe
2024-01-29T08:12:13.667Z modbus-serial modbusSerialErrorHandling:Error Resource temporarily unavailable Cannot lock port
2024-01-29T08:12:13.668Z contribModbus:queue:core queue serial lock command node name: Modbus RS485 id: e39ff271e27b2afe
[... 563 lines cut ...]
        "unit_id": "",
        "commandDelay": "10",
        "clientTimeout": "1000",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": false,
        "showWarnings": true,
        "showLogs": true
    }
]
```

### What did you expect to happen?

Reconnect immediately.

### Other Information

`/dev/ttyRS485` is a symlink to `/dev/ttyUSB0` which connects to a typical FTDI RS-485 interface.

NodeRED v3.1.0, NodeJS v18.18.2, Linux v6.1.71 (NixOS)

> Which type of OPC UA Server are you connected to?

Not sure what this module has to do with OPC UA...?

biancode, 2024-02-04:
Hi, port closing problems are mostly in the combination of serialport as the underlying lib and the nodejs version, which has a very strict binding between both. Some OS' have a some lock time on serial ports or TTY's. That is the reason to have some reconnect timeout in the config node for control.

[image]

serpent213, 2024-02-13:
I don't know, I'm running the default settings as in your screenshot. Looks fine so far in production, redeploying only single nodes now as necessary. Will keep an eye on it.

Feel free to close this issue. Also happy to help with debugging or testing.

closed 2024-05-15 by github-actions[bot]