BiancoRoyal/node-red-contrib-modbus #444: Help: new install is not working

Tonio16, 2024-02-18:
### Which node-red-contrib-modbus version are you using?

5.30.0

### What happened?

Hello

I was using node-red and your node-red-contrib-modbus on a old Pi to read/write the data on ventillation system, Drexel aerosmart L. It was working since years.

I move to a VM with node-red and so I installed the last version of the node, 5.30.0.
On this new install, I have always the same error messages. I was stupide enough to update my pi and now it's also not working on. I don't understand what I am doing wrong there.

`"Client -> fsm broken state after failed Get More About It By Logging  Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 1"`
and
`"Client -> fsm failed state after sending Get More About It By Logging  Serial@/dev/ttyUSB0:19200bit/s default !`

after a while I have:
`"Error: Port Not Open"`
`Client -> fsm init state after reconnecting  Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 1`

Here my settings
[setting_ModbusRTU](https://github.com/BiancoRoyal/node-red-contrib-modbus/assets/48793860/61db9ee7-d33f-45e7-bc78-870c8af22b36)

And the flow
[image]

### Server

Modbus-Server Node

### How can this be reproduced?

Update from a quite old node version (2020?) to the last version

### What did you expect to happen?

No change of behavior within the node versions.

### Other Information

My ventilation system is based on a RS232 Modbus RTU.
It's uint32 or int32. I have to make a different request for each register and to wait at least 300ms between each enquery. I image that the new queueing system is of interest in this case.

Tx

Antoine

biancode, 2024-03-22:
With a VM or Docker it is possible to build some with the old versions to freeze that point of a working version. We test as good as we can with hardware and software and we have many tests in node-red as flows to check if all works well. If all is fine after two running tests and a whole coverage check testing we release the new versions. We testing manual with serial ports on devices and VMs as well as Docker Containers.

Some ideas:
Did you check the Node.js version and all versions like serialport, node-red, modus-serial to be compatible?

It is not just the contrib package what could make the problem. All together is very complex under the hood especially for serial connections. There are maybe new rights in Linux to give like dailout group or others to get connections on serial devices. Did you check that points, too?

Tonio16, 2024-03-22:
Hello

I found a post that describes how to install older version. With the 5.14.0, it's working fine.

I will keep as it is starting from now.

Thanks for your answer

Antoine

closed 2024-03-22 by Tonio16