BiancoRoyal/node-red-contrib-modbus #448: Modbus Flex FC - Loading Codes Issue

innoagg, 2024-03-04:
### Which node-red-contrib-modbus version are you using?

5.30.0

### What happened?

Trying to use the Modbus Flex FC node. I want to define my own map file and load it into the UI. I am not having success. The drop down never populates.

### Server

None/This is related to a node that doesn't connect to a server

### How can this be reproduced?

Ensure a codes.json file exists in the search path defined in the node:

[image]

A valid file exists but the function code selection does not populate.

biancode, 2024-03-22:
@grewek can you check that for us?

grewek, 2024-03-26:
Sorry for the late reply, will take a look at it early next week.

Only thing you could try is to use a absolute path instead of a relative maybe that will help for now :)

grewek, 2024-03-27:
Hi,

here is a little workaround for the issue, i will of course fix the issue as soon as i can.

Put your `codes.json` inside `node-modules/node-red-contrib-modbus/extras/customMaps/` then inside
NODE-Red change the path from `./node_modules/node-red-contrib-modbus/extras/argumentMaps/defaults/` 
to `extras/argumentMaps/customMaps/`

innoagg, 2024-04-08:
Thanks for the above. I did try this with no success. Permissions are correct. Perhaps some UI feedback when you are able to push a fix so we can see what errors are raised if any.

closed 2024-07-24 by github-actions[bot]