BiancoRoyal/node-red-contrib-modbus #450: Need to replace vulnerable vm2 library

wz2b, 2024-03-05:
### Which node-red-contrib-modbus version are you using?

5.30.0

### What happened?

When you install node-red-contrib-modbus npm reports:

  The library contains critical security issues and should not be used for production! The maintenance of the project has been discontinued. Consider migrating your code to isolated-vm.

### Server

Modbus-Server Node

### How can this be reproduced?

Install the package from the command line (using npm) and watch the output

### What did you expect to happen?

I expect to be able to install the package without any severity=critical security warnings

### Other Information

This was reported previously but closed by the bot due to inactivity. There are previous CVEs out there that all say the problem is with vm2 3.9.18 but this is installing 3.9.19 and I still get the warning.  https://www.npmjs.com/package/vm2 suggests migrating from vm2 to isolated-vm

biancode, 2024-03-22:
Feel free to support us to solve all that issues!

biancode, 2024-03-22:
A switch over to the node-red used vm is possible, but has some issues to test if vm can do the same work.

S474N, 2024-05-04:
Still deprecated vm2:

```
2024-05-04T18:07:08.610Z Install : node-red-contrib-modbus 5.31.0

2024-05-04T18:07:09.942Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-modbus@5.31.0
2024-05-04T18:07:10.138Z [err] npm
2024-05-04T18:07:10.138Z [err]  WARN config production Use `--omit=dev` instead.
2024-05-04T18:07:15.168Z [err] npm WARN deprecated vm2@3.9.19: The library contains critical security issues and should not be used for production! The maintenance of the project has been discontinued. Consider migrating your code to isolated-vm.
2024-05-04T18:07:16.025Z [out] 
2024-05-04T18:07:16.025Z [out] added 34 packages in 6s
2024-05-04T18:07:16.031Z rc=0
```

biancode, 2024-05-17:
#461 - comes soon with 5.40.+

closed 2024-08-24 by github-actions[bot]