BiancoRoyal/node-red-contrib-modbus #460: Simulating delay in modbus-flex-server

niranjanshukla, 2024-05-03:
### Describe the requested feature

This is more of a question for now, on how to simulate a delay i n modbus-flex-server.

I have modified the underlying library @open-p4nr/modbus-serial/servers/servertcp.js and added a delay in sending a response

```
sock.on("data", function(data) {
 ...
setTimeout(
                        _parseModbusBuffer.bind(this,
                            requestBuffer,
                            vector,
                            serverUnitID,
                            sockWriter,
                            options
                        ),
                    **360**
                    );
}
```

The issue is, even as I add the above change, and when I connect modbus-flex-getter to modbus-flex-server, modbus flex server ends up sending less number of response objects than it would, if there were no delay set above.

So it looks like the TCP connection is alive only for a certain time. Where may I possibly find the parameter in modbus-flex-server.js / modbus-flex-getter.js or in one of the core modules to increase the connection alive time, so all responses are sent with the above delay?

(On that note, my initial approach was to utilise the responseDelay node param in Flex Server node, but in its implementation, modbus-flex-server.js initialises but never used responseDelay).

### Motivation

Simulating a delay in modbus flex server response

biancode, 2024-05-04:
The package has a split for that and we keep you informed. Give us the chance to switch the vm2 to another first.

niranjanshukla, 2024-05-05:
Thanks @biancode . On a related issue, I have a question - Are _multiple_ modbus-Flex-Server nodes supported in the same flow (and multiple modbus-flex-getter nodes in another flow, reading from those servers)? In my workspace, when I try the following configuration, I do not receive a response on the second modbus-flex-getter node.

**Flow 1:**
modbus-flex-connector1 > modbus-flex-getter1 (using modbus client for server 0.0.0.0, port 502)
modbus-flex-connector2> modbus-flex-getter2(using modbus client for server 0.0.0.0, port 503) [no response received here]

**Flow 2:**
modbus-flex-connector1 > modbus-flex-witer1 (using modbus client for server 0.0.0.0, port 502) + modbus-flex-server1 (serving on host 0.0.0.0, port 502)
modbus-flex-connector2> modbus-flex-writer2(using modbus client for server 0.0.0.0, port 503) + modbus-flex-server2 (serving on host 0.0.0.0, port 503)

When I see Flow 2, I observe both flex writer nodes writing to addresses. But when I see Flow 1, I do not observe flex-getter2 receiving any response.

closed 2024-08-16 by github-actions[bot]