BiancoRoyal/node-red-contrib-modbus #464: modbus server , address set error while using , and I fixed it

wei1224hf, 2024-05-16:
### Which node-red-contrib-modbus version are you using?

/home/gene/nodejs/userDir1881

### What happened?

using modbus server , 
inject data from input , with funcion content:

msg.payload = { 'value': 1, 'register': 'holding', 'address': 1, 'disableMsgOutput': 0 };
return msg;

after sending this to modbus-server node , the address 1 did not change , but the address 8 changed.

### Server

Modbus-Server Node

### How can this be reproduced?

deploy a modbus server node , send data :
msg.payload = { 'value': 1, 'register': 'holding', 'address': 1, 'disableMsgOutput': 0 };

### What did you expect to happen?

using another software ModScan32.exe , monitoring data , 
the address 1 data of the modbus server from node-red should changed to 1, 
but it did not , address 8 changed to 1 instead.

### Other Information

I have already fixed this bug by 

modbus-server-core.js , line 130:

de.biancoroyal.modbus.core.server.writeToModbusBuffer = function (node, msg) {
  switch (msg.payload.register) {
    case 'holding':
      node.modbusServer.holding.writeUInt16BE(msg.bufferPayload, msg.bufferAddress/4)
      break
    case 'coils':
      //console.debug(msg);
      function setBit(buffer, i, bit, value){
        if(value == 0){
          buffer[i] &= ~(1 << bit);
        }else{
          buffer[i] |= (1 << bit);
        }
      }
      const address = msg.bufferAddress/8;
      const idx = Math.floor(address/8) ;
      const idx2 = address%8;
      setBit(node.modbusServer.coils,idx,idx2,msg.bufferPayload);
      //node.modbusServer.coils.writeUInt8(msg.bufferPayload, msg.bufferAddress)
      //console.debug(node.modbusServer.coils);
      break
    case 'input':
      node.modbusServer.input.writeUInt16BE(msg.bufferPayload, msg.bufferAddress/4)
      break
    case 'discrete':
      const address = msg.bufferAddress/8;
      const idx = Math.floor(address/8) ;
      const idx2 = address%8;
      setBit(node.modbusServer.discrete,idx,idx2,msg.bufferPayload);
      //node.modbusServer.discrete.writeUInt8(msg.bufferPayload, msg.bufferAddress)
      break
    default:
      return false
  }
  return true
}

biancode, 2024-06-23:
Hi, can I ask you to test again with version 5.40+?

closed 2024-10-24 by github-actions[bot]

fkroepfl, 2025-01-14:
@wei1224hf @biancode 

Hello, as a complete beginner I obviously ran into the problem described here without realising it. I am struggling with the problem described above that IMO the address index is not being processed correctly.

Working with 5.43.0, I write with the values 0, 2, 4 for ‘startAddress’ as follows:

// Construct the Modbus message payload
msg.payload = {
    value: hexByteArray, // Pass the raw bytes directly
    register: targetRegister, // Specify the target register (‘input’ or ‘holding’)
    address: startAddress // Specify the start address
};

See the excerpt from the buffer payload after the operation and for comparison how the payload should look in my opinion.

Input.Register Buffer is actual:
-  {"payload":[**9,31,0,0**,0,0,0,0,0,0,0,0,0,0,0,0,**9,34,0,0**,0,0,0,0,0,0,0,0,0,0,0,0,**9,67,0,0**]}

Input.Register Buffer, IMHO, should be:
- {"payload":[**9,31,0,0**,**9,34,0,0**,**9,67,0,0**,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}

My mistake or system bug?

Many thx  in advance. 🤗