BiancoRoyal/node-red-contrib-modbus #469: Modbus ASCII over TCP/IP not working

mrfy, 2024-06-18:
### Which node-red-contrib-modbus version are you using?

5.31.0

### What happened?

I am testing this library to read the Delta PLC DVP14SS2 controller, which operates in Modbus ASCII mode. I communicate with the controller using the USR-TCP232-410s RS232 / RS485 Ethernet converter. The Delta PLC controller is connected to the RS232 input of my converter.

I am reporting an issue because I am unable to read the controller's registers; in Node-Red, I receive the message 'Error: Timed Out', while comunication is set as TCP.

[image]

 For comparison, using the Modbus Poll program, I am able to read the controller correctly. For this configuration:

[image]

When I connect the serial port from my laptop directly to the PLC controller and change the configuration in Node-Red to use the COM port and set the connection mode to ASCII, I can read the registers without any problems.

[image]

I searched the forum but found two issues related to ASCII communication, but issues are related to serial connection. It appears that ASCII is unsupported in TCP mode.

### Server

Other/External server

### How can this be reproduced?

Trying to read modbus ASCII device using TCP/IP connection.

### What did you expect to happen?

Correctly read registers using TCP Modbus ASCII

biancode, 2024-06-23:
Hi, if the serial settings of the Modbus poll are the defaults, then you need to set them as in Modbus poll via the Serial Expert  configuration type of the config node. Give it a try with the 7 data Bits, even parity and 1 stop Bit

mrfy, 2024-06-24:
> Hi, if the serial settings of the Modbus poll are the defaults, then you need to set them as in Modbus poll via the Serial Expert configuration type of the config node. Give it a try with the 7 data Bits, even parity and 1 stop Bit

Thank you for response but I think we little misnundersood :)

Please take a look on Modbus Poll Read/Write Definition - I can correctly read modbus registers:
[image]

Similar configuration in Node Red -  I cannot:

[image]
[image]

I hope that screens give you full view on the situation :)

mrfy, 2024-06-25:
Hello, I have do more tests. And I think I know where is the issue. When I'm sending raw HEX modbus frame using TCP connection in node-red I have response:
[image]

Function, and response:

[image]

biancode, 2024-07-24:
If you switch on more debugging in logs see Wiki Article [Debug]( https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG) and start node-red with parameter -v (verbose) - then you should see what we send and what is the problem

closed 2024-10-24 by github-actions[bot]