BiancoRoyal/node-red-contrib-modbus #474: Modbus-Flex-Connector error "setting 'nodeStatus'"

DyachenkoAlexey, 2024-06-29:
### Which node-red-contrib-modbus version are you using?

5.40.0

### What happened?

After updating node-red-contrib-modbus, Modbus-FlexConnector began to display the message "TypeError: Cannot set properties of undefined (setting 'nodeStatus')". Adding the "nodeStatus" parameter to the input message for configuring the port did not lead to positive results.

### Server

Modbus-Server Node

### How can this be reproduced?

Send new port settings to the input in accordance with the example to the node.

### What did you expect to happen?

The settings were applied as in previous versions. But now they don't work.

MaryamJalil, 2024-07-01:
Hi @DyachenkoAlexey , Can you share your example flow for this issue in the comment too

DyachenkoAlexey, 2024-07-01:
Hi. My solution is to go back to the previous version. My project requires changing the connection settings via dashboard.

biancode, 2024-07-23:
I checked it and there was a bigger mistake from a merge - it will be fixed soon and brings some new tests and improvements.

biancode, 2024-07-23:
fixed with version 5.41.0

closed 2024-10-24 by github-actions[bot]