BiancoRoyal/node-red-contrib-modbus #476: Modbus Flex Connector not setting unit id of modbus client

schollsi, 2024-07-04:
### Which node-red-contrib-modbus version are you using?

5.31.0

### What happened?

I want to read modbus tcp data from different modbus devices. I created a subflow with environment variables to set the host-ip, the port, the unit id and the address to read:

[image]

In the "Server Config" function node the modbus client is defined and provided to the Modbus Flex Connector:

```javascript
msg.payload = {
    connectorType: 'TCP',
    tcpHost: env.get("modbus_host"),
    tcpPort: env.get("modbus_port"),
    unit_id: env.get("modbus_unitid"),
};
return msg;
``` 

In the Modbus Flex Conncetor i created a server with the name "Default_Client". The server has default values:

[image]

In the "Define Address" function node the address to read is set and provided to the Modbus Flex Getter:

```javascript
msg.payload = {
    fc: 3,
    address: env.get("modbus_address"),
    quantity: 1
};

return msg
``` 

The Server Config node is triggered once after 1 second. The Define Address node is triggered once after 1 second, then every 1 seconds.

I created a flow with two instances of the subflow. One instance for a modbus connection with unit id 1, the second for another modbus connection to a different device with unit id 10:

[image]

[image]

 Whenn running the flow, I only get values from the first instance with unit id 1, which matches the default value of the "Default_Client". When changing the defualt Unit-Id of the "Default_Client" to 10, I only get values from the second device, which has unit id 10. It seems like the Modbus Flex Connector is not overwriting the default Unit-Id with the value it get's from the "Server Config" node but keeps the default value of the "Default_Client".

### Server

Other/External server

### How can this be reproduced?

- Create two instances of the ReadModbus-Subflow
- First instance with modbus_unitid = 1
- Second instance with modbus_unitid ~= 1
- Connect the outputs of both instances to debog nodes
- Only the subflow with modbus_unitid = 1 delivers values

```javascript
[
    {
        "id": "92094e3b8865d186",
        "type": "subflow",
        "name": "ReadModbus",
        "info": "",
        "category": "",
        "in": [],
        "out": [
[... 274 lines cut ...]
        "type": "debug",
        "z": "33fbc91930097586",
        "name": "debug 4",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "x": 620,
        "y": 240,
        "wires": []
    }
]
``` 

### What did you expect to happen?

The modbus clients in both instances are set with the right Unit-Id and both subflows deliver values to the debug nodes.

### Other Information

Node-RED version: 4.0
NodeJS version: 20.15.0

biancode, 2024-07-23:
it has to be msg.payload.unitId

biancode, 2024-07-23:
node.unit_id is the property of the node itself later that will be set from the msg.payload.unitId if it exists in the payload and it is a number ready to be parsed as Integer

schollsi, 2024-07-23:
Thanks for your suggestion. The Unit-Id was misspelled in my approach, you are right. However, wheat really solved the issue for me was to ad the unitId to the payload that is sent to the Modbus-Flex-Getter node instead of the payload that is sent to the Modbus-Flex-Connector node.

closed 2024-10-24 by github-actions[bot]