BiancoRoyal/node-red-contrib-modbus #477: FC20 Read File Record doesn't work

me21, 2024-07-09:
### Which node-red-contrib-modbus version are you using?

5.40.0

### What happened?

I'm trying to read file record using code 20 (14h). I use Modbus Flex-FC node for that. The device I'm querying is a real external device connected to a serial port over Modbus RTU.
I connected an oscilloscope to the serial lines and I can see the request and the response. But the response isn't correctly parsed.
Request: F7 14 07 06 00 03 00 00 00 04 59 3B
Response: F7 14 0A 09 06 00 00 00 00 00 00 00 00 A2 3A

### Server

Other/External server

### How can this be reproduced?

I think it can be reproduced by trying to read file record from any server which supports that feature.

### What did you expect to happen?

A response should be correctly parsed.

### Other Information

The flow looks like this:

[image]

I can provide any additional information you may need.

biancode, 2024-07-23:
File record is special and not supported out-of-the-box - that why I see this more as a feature request at the moment

me21, 2024-07-23:
I have a patch which I can share. But it's for @open-p4nr/modbus-serial package which is downloaded from a private repository if I'm correct. How do I submit it?

closed 2024-10-24 by github-actions[bot]

me21, 2024-10-24:
@biancode please feel free to reopen it if it's not really completed.