BiancoRoyal/node-red-contrib-modbus #480: modbus-flex-getter crashes Node Red v 4.0.2

ilexpl01, 2024-07-21:
### Which node-red-contrib-modbus version are you using?

5.40.0

### What happened?

Once a day my dockerized Node Red crashes due to modbus-flex-getter failure

### Server

Other/External server

### How can this be reproduced?

Log below:

21 Jul 19:08:06 - [warn] [modbus-flex-getter:f794789b3dbe6dd3] Modbus Failure On State sending Get More About It By Logging

21 Jul 19:09:05 - [red] Uncaught Exception:

    at boundsError (node:internal/buffer:88:9)

    at Buffer.readUInt16BE (node:internal/buffer:334:5)

    at _readFC3or4 (/data/node_modules/@open-p4nr/modbus-serial/index.js:160:26)

    at ModbusRTU._onReceive (/data/node_modules/@open-p4nr/modbus-serial/index.js:560:17)

    at TcpPort.emit (node:events:519:28)

    at Socket.<anonymous> (/data/node_modules/@open-p4nr/modbus-serial/ports/tcpport.js:93:24)

    at Socket.emit (node:events:519:28)

    at addChunk (node:internal/streams/readable:559:12)

    at readableAddChunkPushByteMode (node:internal/streams/readable:510:3)

    at Socket.Readable.push (node:internal/streams/readable:390:5)

    at TCP.onStreamRead (node:internal/stream_base_commons:191:23)

21 Jul 19:10:45 - [info]

biancode, 2024-07-23:
A case like this needs much more information - NodeJS version, where does it run and how to reproduce it every time. Feel free to write a test and send it as a PR here! Then we can investigate easier and faster.

ilexpl01, 2024-07-23:
Hi.

 

Please have a look below:

nodered

date

stream

content

2024/07/23 09:29:38

stdout

23 Jul 09:29:38 - [info] Loading palette nodes

2024/07/23 09:28:33

stdout

23 Jul 09:28:29 - [info] Linux 4.4.302+ x64 LE

2024/07/23 09:28:33

stdout

23 Jul 09:28:29 - [info] Node.js  version: v20.15.0

2024/07/23 09:28:33

stdout

23 Jul 09:28:29 - [info] Node-RED version: v4.0.2

2024/07/23 09:28:33

stdout

	

2024/07/23 09:28:33

stdout

===================

2024/07/23 09:28:33

stdout

Welcome to Node-RED

2024/07/23 09:28:33

stdout

	

2024/07/23 09:28:29

stdout

23 Jul 09:28:29 - [info] 

2024/07/23 09:14:12

[... 240 lines cut ...]
stdout

23 Jul 09:12:05 - [warn] [modbus-flex-getter:1ab9b4b5390bc392] Modbus Failure On State sending Get More About It By Logging

2024/07/23 09:12:08

stdout

23 Jul 09:12:05 - [warn] [modbus-flex-getter:bb60410a899d0402] Modbus Failure On State sending Get More About It By Logging

2024/07/23 09:12:08

stdout

23 Jul 09:12:05 - [warn] [modbus-flex-getter:2fe658c0d18ba8ff] Modbus Failure On State sending Get More About It By Logging

2024/07/23 09:12:05

stdout

23 Jul 09:12:05 - [warn] [modbus-flex-getter:69e4934e7c166455] Modbus Failure On State sending Get More About It By Logging

 

I can not reproduce an error on request :/

ilexpl01, 2024-07-26:
Hi,

Upgrade completed:

26 Jul 13:52:36 - [info] Upgrading module: node-red-contrib-modbus to version: 5.41.0

Current setup:

Welcome to Node-RED
===================

26 Jul 13:53:55 - [info] Node-RED version: v4.0.2
26 Jul 13:53:55 - [info] Node.js  version: v20.15.0
26 Jul 13:53:55 - [info] Linux 4.4.302+ x64 LE

We will see if it helps :)

biancode, 2024-07-26:
I'd expect it doesn't because the modbus-serial dependency still the same for this update and it looks like it is a problem in that package, what we can check in August.

ilexpl01, 2024-07-26:
Will make a buzz in August so :)

ilexpl01, 2024-07-26:
You were right:

26 Jul 17:34:17 - [warn] [modbus-flex-getter:503a2248aafc2d3d] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:ed290d67cc9457b4] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:91ce210a1b743790] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:6bccb3d1b377d72a] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:c97c628501952fec] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:0dc50987c4e6ea6a] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:b2728bde45725482] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:aea8dc8a5d4c631f] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:a9df80f99f032dcd] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:b2ccadaf53c091f5] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:34:17 - [warn] [modbus-flex-getter:3212baaad1b117cb] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:69e4934e7c166455] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:2fe658c0d18ba8ff] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:bb60410a899d0402] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:1ab9b4b5390bc392] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:fae969e560e34da7] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:34abd5f2c5e9a5e9] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:63f2c18edad16ca5] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:22f91f8503cc96b9] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:67d70776fa66258e] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:03da07f1c05ec6f0] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:1776995f76909225] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:35:17 - [warn] [modbus-flex-getter:f794789b3dbe6dd3] Modbus Failure On State sending Get More About It By Logging
26 Jul 17:37:15 - [red] Uncaught Exception:
26 Jul 17:37:15 - [error] RangeError: The value of "offset" is out of range. It must be >= 0 and <= 7. Received 9
    at boundsError (node:internal/buffer:88:9)
    at Buffer.readUInt16BE (node:internal/buffer:334:5)
    at _readFC3or4 (/data/node_modules/@open-p4nr/modbus-serial/index.js:160:26)
    at ModbusRTU._onReceive (/data/node_modules/@open-p4nr/modbus-serial/index.js:560:17)
    at TcpPort.emit (node:events:519:28)
    at Socket.<anonymous> (/data/node_modules/@open-p4nr/modbus-serial/ports/tcpport.js:93:24)
    at Socket.emit (node:events:519:28)
    at addChunk (node:internal/streams/readable:559:12)
    at readableAddChunkPushByteMode (node:internal/streams/readable:510:3)
    at Socket.Readable.push (node:internal/streams/readable:390:5)
    at TCP.onStreamRead (node:internal/stream_base_commons:191:23)

Till August so.

biancode, 2024-07-27:
It would be very helpful if you could give more information from the [Debug insights](https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG):

`DEBUG=contribModbus*,modbus-serial node-red -v`

biancode, 2024-07-27:
[What is the DEBUG 🐛 environment variable in Node.js, and how to use it?](https://dev.to/aderchox/what-is-the-debug-environment-variable-in-nodejs-and-how-to-use-it-3fal)

ilexpl01, 2024-07-27:
> It would be very helpful if you could give more information from the [Debug insights](https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG):
> 
> `DEBUG=contribModbus*,modbus-serial node-red -v`

I am affraid I have no idea where to place this statement :(

Should this be in NR settings.js file or as an env in NR container config ?

Sorry 4 asking.

hacesoft, 2024-07-30:
To configure logging in Node-RED, you need to modify the settings.js file. This file contains configuration settings for various aspects of Node-RED, including logging. Here's a step-by-step guide:

Step 1: Locate the settings.js File
Location of settings.js:

If you installed Node-RED locally, the file is typically found in the ~/.node-red/ directory.
For Docker setups, the file might be located in a mounted volume where your data is stored.
Open the settings.js file using a text editor.

Step 2: Configure Logging
In the settings.js file, locate the logging section and configure it as needed. Here’s an example configuration:

```
logging: {
    console: {
        level: "info", // Logging levels: fatal, error, warn, info, debug, trace
        metrics: false, // Set to true to enable metrics
        audit: false,  // Set to true to enable audit logs
        handler: function(settings) {
            // Custom handler for processing logs if desired
        }
    }
}
```
Step 3: Parameter Descriptions
level: Sets the minimum level of messages to log. Options include fatal, error, warn, info, debug, and trace. The info level logs most standard messages, while debug and trace provide more detailed debugging information.
metrics: When set to true, Node-RED will collect performance metrics.
audit: When set to true, Node-RED will collect audit logs, which may include details about changes made to the system.
handler: Allows defining a custom function for handling logs. In most cases, the default console handler is used.
Step 4: Save and Restart Node-RED
After editing the settings.js file, restart Node-RED to apply the changes:

Local Installation: If installed locally, stop and restart Node-RED using commands like node-red-stop and node-red-start.
Docker: If using Docker, restart the container with docker restart <container_name>.
Usage in a Node-RED Function Node
To log messages from within a function node, use the following methods:

```
node.log("Informational message");  // Logs at info level
node.warn("Warning message");       // Logs at warn level
node.error("Error message");        // Logs at error level
node.debug("Debug message");        // Logs at debug level
node.trace("Trace message");        // Logs at trace level
```

These messages will be logged to the console according to the set logging level.

For more details, you can refer to the Node-RED documentation on logging: [Node-RED Logging Documentation](https://nodered.org/docs/user-guide/runtime/logging).

ilexpl01, 2024-07-30:
hacesoft - thx 4 the tutorial. I already new the trick with settings.js file but it does output the log to console while I need to get them in a file. I Googled quite some time with no answer found how to achive that :/

hacesoft, 2024-07-30:
I just asked an AI focused on NODE-RED and this is what she told me :):

Yes, you can redirect Node-RED's log output to a file by configuring the logging section in the settings.js file. This is useful for keeping a persistent log of errors, warnings, and other messages for debugging or monitoring purposes. Here's how you can set it up:

Redirecting Logs to a File
To redirect logs to a file, you can modify the logging configuration in the settings.js file as follows:

```
logging: {
    file: {
        level: "info", // Set the desired log level
        metrics: false, // Disable or enable metrics
        audit: false,   // Disable or enable audit logs
        path: "/path/to/your/log/file.log", // Specify the log file path
        // Optional - custom handler
        handler: function(settings) {
            // Custom processing of log messages
        }
    }
}
```
Configuration Options:
level: Specifies the minimum level of logs to record. For instance, setting it to info means it will log info, warn, error, etc.
metrics: Set to true to log performance metrics.
audit: Set to true to log audit events.
path: The full path to the log file where you want to store the logs. Ensure that the directory exists and is writable by the Node-RED process.
Example
If you want to log all messages of level info and above to a file located at /var/log/node-red.log, your configuration would look like this:

```
logging: {
    file: {
        level: "info",
        metrics: false,
        audit: false,
        path: "/var/log/node-red.log"
    }
}
```
Additional Considerations
Log Rotation: Node-RED does not handle log rotation by itself. You may need to use external tools like logrotate on Linux systems to manage the size and rotation of log files.
Permissions: Ensure that the user running Node-RED has write permissions to the directory where the log file is stored.
Custom Handlers: If you want to customize how logs are handled (e.g., adding timestamps, formatting), you can define a custom handler function in the handler property.
Saving and Applying Changes
After editing the settings.js file, save your changes and restart Node-RED to apply the new logging configuration. This will redirect all log messages, including errors and warnings, to the specified file.

hacesoft, 2024-07-30:
I think it is necessary to write /data/log_file.log and the file will be in the root directory. or how do you have a specified storage path...

[image]

And this is where it should be stored...

[image]

We are talking about dockers on Synology NAS

ilexpl01, 2024-07-30:
I have it this way:

[image]

but log_file.log does not show up in folder:

[image]

hacesoft, 2024-07-30:
Refresh the folder, click on another address book and back to the addressee node-red...

ilexpl01, 2024-07-30:
Please, it is not there !

hacesoft, 2024-07-30:
I looked into the configuration file and the second line under the keyword LOGGING says that it only works for consoles.

[image]

But here: https://nodered.org/docs/user-guide/runtime/logging there is a logger at the bottom of the page and it could be used to send data to the log center (an application that needs to be installed on synology). We haven't tried it, but I think it should work. If you start it, let it lead.

biancode, 2024-07-31:
With Docker you can set the -env DEBUG=....

ilexpl01, 2024-07-31:
> With Docker you can [set the -env](https://docs.docker.com/compose/environment-variables/set-environment-variables/) DEBUG=.... (and mostly you have to restart the container if you do it with Portainer or a UI tool)

I tried DEBUG = 1, DEBUG = true, DEBUG = /data/logfile.txt (of course with container restar every change).

No luck, file does not appear. 

Have a look here:

https://nodered.org/docs/getting-started/docker

there is not a word about DEBUG env.

ilexpl01, 2024-07-31:
Env DEBUG = * start did the trick but the output still goes to console, not a file.

ilexpl01, 2024-07-31:
> It would be very helpful if you could give more information from the [Debug insights](https://github.com/BiancoRoyal/node-red-contrib-modbus/wiki/DEBUG):
> 
> `DEBUG=contribModbus*,modbus-serial node-red -v`

Hi, Got some logs. How can I pass them to You Klaus ?

biancode, 2024-08-01:
send it to support at plus4nodered.com via E-Mail

ilexpl01, 2024-08-01:
> send it to support at plus4nodered.com via E-Mail

done

ilexpl01, 2024-08-07:
> send it to support at plus4nodered.com via E-Mail

any update regarding sent logs ?

grewek, 2024-08-07:
> > send it to support at plus4nodered.com via E-Mail
> 
> any update regarding sent logs ?

Yes we received the log and i took a look at the underlying problem if we take a look at the last received message from the modbus device:

```
Buffer <00 0d 00 00 00 07 01 04 0c 43 5e 1b c6>
```

it is indicating that it has read 0x0C(12) Registers so after the 0x0C there should be 12 more bytes following but as we can see there are only ```43 5e 1b c6``` 4 bytes on the message which later down the line leads to a crash because javascript tried to access a value which would be out of bounds. I think this is fixable if i identified the source correctly :)

I need a bit more time to do more research because i want to make sure that i don't break other stuff with the bugfix i have in mind.

grewek, 2024-08-13:
Okay the fix is released in V5.42.0 :)

Hope that this fixes the issue if not please let me know!

ilexpl01, 2024-08-13:
> Okay the fix is released in V5.42.0 :)
> 
> Hope that this fixes the issue if not please let me know!

[image]

Not yet available, will wait :)

ilexpl01, 2024-08-14:
Module upgraded:

[image]

Hope not to have any more crashes of NR.
Need couple of days to judge jour fix @grewek   :)
We are in touch.

valdiks, 2024-08-14:
I have the same issue. I've updated to 5.42.0 but issue persists. Will send debug info later.

grewek, 2024-08-14:
@valdiks yes please send them to support at plus4nodered.com
maybe i missed something.

grewek, 2024-08-15:
@valdiks i looked at the log and this seems not to be related to this issue, please open a new issue with all necessary information.

Thank you!

ilexpl01, 2024-08-24:
No further issues after upgrade to 5.42.0. Thx

closed 2024-08-24 by ilexpl01

zhushaoan, 2024-10-10:
你好，请问在容器里面看到的串口列表是容器里面的还是宿主机的？我看到的是列表是宿主机的，如果使用宿主机的串口发送数据收不到，如果使用容器的串口列表会报7段错误