BiancoRoyal/node-red-contrib-modbus #481: Modbus-Flex-Connector node - actualServiceState

hacesoft, 2024-07-23:
### Which node-red-contrib-modbus version are you using?

5.40

### What happened?

I have a flow where I try to read data from FVE - victron, but that is irrelevant. I want the user to enter his FVE IP address. So I used the Modbus-Flex-Connector node, configured:

msg.payload = {
 'connectorType': 'TCP',
 'tcpHost': '192.168.50.200',
 'tcpPort': '502',
 'unitId': 100
};

return msg;

The node normally connects using the Modbus READ node is read periodically, but the Modbus-Flex-Connector node throws an error message after connection:

23/07/2024 11:39:15 node: connection3
msg: error
"TypeError: Cannot read properties of undefined (reading 'actualServiceState')"

At the same time, everything works, what kind of message is this, what did I miss?

### Server

Other/External server

### How can this be reproduced?

[
    {
        "id": "5f2008296b1f5903",
        "type": "tab",
        "label": "Flow 3",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "099b50a7f3c0eefb",
        "type": "modbus-read",
        "z": "5f2008296b1f5903",
        "name": "Modbus Read",
        "topic": "phase1",
        "showStatusActivities": false,
        "logIOActivities": false,
        "showErrors": false,
        "showWarnings": true,
        "unitid": "100",
        "dataType": "InputRegister",
        "adr": "817",
        "quantity": "1",
        "rate": "1",
        "rateUnit": "s",
        "delayOnStart": false,
        "startDelayTime": "",
        "server": "6c801ec0cc108747",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "x": 710,
        "y": 200,
        "wires": [
            [
                "9100e5a706c35769"
            ],
            []
        ]
    },
    {
        "id": "9100e5a706c35769",
        "type": "debug",
        "z": "5f2008296b1f5903",
        "name": "debug 36",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "payload",
        "statusType": "auto",
        "x": 900,
        "y": 200,
        "wires": []
    },
    {
        "id": "05c558792be643c3",
        "type": "function",
        "z": "5f2008296b1f5903",
        "name": "connect",
        "func": "msg.payload = {\n    'connectorType': 'TCP', \n    'tcpHost': '192.168.50.200', \n    'tcpPort': '502',\n    'unitId': 100\n\n}; \n\nreturn msg;\n\n/*\n\nmsg.payload.tcpHost || node.tcpHost\nmsg.payload.tcpPort || node.tcpPort\nmsg.payload.tcpType || node.tcpType\nmsg.payload.unitId || node.unit_id\nmsg.payload.commandDelay || node.commandDelay\nmsg.payload.clientTimeout || node.clientTimeout\nmsg.payload.reconnectTimeout| || node.reconnectTimeout\n\n*/",
        "outputs": 1,
        "timeout": "",
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 540,
        "y": 280,
        "wires": [
            [
                "7f91a08aeb455892",
                "91edcba2daa6fca6"
            ]
        ]
    },
    {
        "id": "7f91a08aeb455892",
        "type": "debug",
        "z": "5f2008296b1f5903",
        "name": "debug 37",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "payload",
        "statusType": "auto",
        "x": 700,
        "y": 340,
        "wires": []
    },
    {
        "id": "b8e5428f00bea9a3",
        "type": "inject",
        "z": "5f2008296b1f5903",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 380,
        "y": 280,
        "wires": [
            [
                "05c558792be643c3"
            ]
        ]
    },
    {
        "id": "91edcba2daa6fca6",
        "type": "modbus-flex-connector",
        "z": "5f2008296b1f5903",
        "name": "connection3",
        "maxReconnectsPerMinute": 4,
        "emptyQueue": false,
        "showStatusActivities": true,
        "showErrors": true,
        "server": "6c801ec0cc108747",
        "x": 710,
        "y": 280,
        "wires": [
            [
                "f3b6b05fcf177f4b"
            ]
        ]
    },
    {
        "id": "f3b6b05fcf177f4b",
        "type": "debug",
        "z": "5f2008296b1f5903",
        "name": "debug 38",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "payload",
        "statusType": "auto",
        "x": 880,
        "y": 280,
        "wires": []
    },
    {
        "id": "6c801ec0cc108747",
        "type": "modbus-client",
        "name": "xxx",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "failureLogEnabled": false,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": false,
        "showErrors": false,
        "showWarnings": true,
        "showLogs": true
    }
]

### Other Information

I am using nodered 4.0.2 version.

biancode, 2024-07-23:
fixed with #476

hacesoft, 2024-07-23:
I don't know what I'm doing wrong all the time. I rewrote the input function so that it supplies node connection parameters, it is the first message in debug, even so, node connection returns the error but continues...

[image]

biancode, 2024-07-23:
please test with the new version 5.41.0 - I was not able to reproduce it with the given flow and it should be fixed. Otherwise I need to learn more how to reproduce it or you could write a test for it in the package e2e tests.

hacesoft, 2024-07-23:
I also tried running the flow shown in the picture on a new instance of node-red, where there are only the nodes shown in the picture. And with node-red-contrib-modbus version 5.40. it reported the same errors there too, I didn't try it primo on:

https://www.victronenergy.com/media/pg/Cerbo_GX/en/index-en.html

There is also node-red, but some old version.

Anyway, version 5.41 node-red-contrib-modbus solved the problem, even debug38 returns some kind of message :).

Thanks a lot for the quick fix...

[image]

Strunck, 2024-07-25:
Thank you. Error message is gone!

hacesoft, 2024-07-28:
Have a nice day,
please, how can it be recognized that the device has connected and is communicating with the given IP address. According to the debug35 output, the response is exactly the same, with a physical device at 192.168.50.200 and nothing at 192.168.60.200. The answer is the same....

[image]

closed 2024-10-27 by github-actions[bot]