BiancoRoyal/node-red-contrib-modbus #483: Modbus-Flex-Connector node indikace stavu služby

hacesoft, 2024-07-30:
### Which node-red-contrib-modbus version are you using?

5.41

### What happened?

Unable to verify current service status. Even if I am disconnected from the TCP communication with the inverter or it is connected or connected, the output from the Modbus-Flex-Connector node is not indicated in any way.

### Server

Other/External server

### How can this be reproduced?

node-red 4.02 drives me to the NAS synology version in docker, and I connect to CERBO (communication device for victron inverters), although it is on another VLAN, but that is a network solution. It is enough to fill the node function with data and send it to the Modbus-Flex-Connector node, connect debug to the output and turn on the output to the goat and you can see that no message comes when disconnecting, no message during connection either, the only thing that comes is a message after connecting to given sent configuration. If you send your configuration, you need another IP address where there is no HW, so a message will come saying that the device did not connect. I only know the connection based on the need for current and data from the modbus-read node.

### What did you expect to happen?

I expected that when disconnecting, a message would come saying that I am disconnected from the TCP server, or a message would come, I am currently trying to connect to the given TCP configuration, or I was able to connect with the given IP address to the given port with this ID

grewek, 2024-08-29:
I took a look at the issue and i can reproduce it, but right now i am not sure what changes would be necessary
to implement the proposed feature. I will do my best to solve this issue but can't make any promise when it will
actually happen! Time is a little bit short these days!

Best Regards,

Kay

hacesoft, 2024-09-28:
Good day, although it's not a 100% solution, but somehow I got hung up on the absence of the connection status check. Here it works reliably :).
Flow works like this: I have a second injector that goes to the Check SN function, there is data for reading the SN system. If the function succeeds, I have a number other than 0. And thus I am connected. If I have a bad IP address or the device is not connected to modbus, the check connection function checks whether I have a valid time stamp, this is so that I receive messages, if I have 4 consecutive valid time stamps, then I am not connected, otherwise the calculator resets and we connected to the IP address :).

`[
    {
        "id": "6f269dbc176339b8",
        "type": "tab",
        "label": "Flow 2",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "38f637084b7cd43f",
        "type": "modbus-flex-connector",
        "z": "6f269dbc176339b8",
        "name": "connection",
        "maxReconnectsPerMinute": 4,
        "emptyQueue": false,
        "showStatusActivities": false,
        "showErrors": false,
        "server": "6c801ec0cc108747",
        "emptyMsgOnFail": false,
        "configMsgOnChange": false,
        "x": 550,
        "y": 320,
        "wires": [
            [
                "bd7406b0bf6f2069"
            ]
        ]
    },
    {
        "id": "66960bd8137f9326",
        "type": "function",
        "z": "6f269dbc176339b8",
        "name": "connect",
        "func": "//*****************************************************************************\n//\n// File Name        : N/A\n// Title            : N/A\n// Author           : http://www.prochazka.zde.cz -> hacesoft 2024\n// Created          : 20-07-2024, 17:01\n// Revised          : 25-07-2024, 19:42\n// Version          : 1.2\n// Target Platform  : Node-RED\n//\n// This code is distributed under the GNU Public License\n// Všechny informace jsou zahrnuty pod GPL licenci, pokud není explicitně uveden jiný typ licence.\n// Používání těchto stránek a produktů ke komerčním účelům lze jen se souhlasem autora.\n// Všechna práva vyhrazena (c) 1997 - 2024 hacesoft.\n//\n//*****************************************************************************\n// Debugging flag\nconst debug = false;  // Nastavte na true pro zapnutí debug zpráv\n\n//definice globalních funkci\nconst fGetConfigProperty = global.get('getConfigProperty');\nconst fSetConfigProperty = global.get('setConfigProperty');\n\nconst tcpHost = fGetConfigProperty(\"connectorConfig.tcpHost\") || 0; //dotaz na data z globalni strukturu\nconst tcpPort = fGetConfigProperty(\"connectorConfig.tcpPort\") || 0; //dotaz na data z globalni strukturu\nconst unitId = fGetConfigProperty(\"connectorConfig.unitId\") || 0; //dotaz na data z globalni strukturu\n\nnode.debug(\"debug message\");\n\nmsg.payload = {\n    'connectorType': \"TCP\",\n    'tcpHost': tcpHost,\n    'tcpPort': tcpPort,\n    'unitId': unitId,\n    'modbusResponse.status': 0\n};\n\n \n\nreturn msg;\n\n/*\nnode.log(\"Informational message\");  // Logs at info level\nnode.warn(\"Warning message\");       // Logs at warn level\nnode.error(\"Error message\");        // Logs at error level\nnode.debug(\"Debug message\");        // Logs at debug level\nnode.trace(\"Trace message\");        // Logs at trace level\n*/\n",
        "outputs": 1,
        "timeout": "",
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 380,
        "y": 320,
        "wires": [
            [
                "38f637084b7cd43f"
            ]
        ]
    },
    {
        "id": "95392e1b35b58aaa",
        "type": "delay",
        "z": "6f269dbc176339b8",
        "name": "500ms",
        "pauseType": "delay",
        "timeout": "500",
        "timeoutUnits": "milliseconds",
        "rate": "1",
        "nbRateUnits": "1",
        "rateUnits": "second",
        "randomFirst": "1",
        "randomLast": "5",
        "randomUnits": "seconds",
        "drop": false,
        "allowrate": false,
        "outputs": 1,
        "x": 230,
[... 201 lines cut ...]
        "parallelUnitIdsAllowed": false,
        "showErrors": false,
        "showWarnings": true,
        "showLogs": true
    },
    {
        "id": "ba1c560b4a2f6b32",
        "type": "ui_group",
        "name": "TCP",
        "tab": "421891c215fdf211",
        "order": 2,
        "disp": true,
        "width": "6",
        "collapse": true,
        "className": ""
    },
    {
        "id": "421891c215fdf211",
        "type": "ui_tab",
        "name": "Config",
        "icon": "settings_input_component",
        "disabled": false,
        "hidden": false
    }
]`

closed 2024-12-29 by github-actions[bot]