BiancoRoyal/node-red-contrib-modbus #487: Disabling one modbus-read node will move other nodes to "closed" state

nagisa, 2024-08-03:
### Which node-red-contrib-modbus version are you using?

5.41.0

### What happened?

I have a couple `modbus-read` nodes that refer to the same server. Usually this works quite alright.

However, when modifying flows I tend to enable and disable nodes or groups of nodes sometimes. When I disable a `modbus-read` node, I notice that all the other nodes that associate themselves with the same `modbus-server` will move to a "closed" state and will stop reading from the server indefinitely (until node-red is restarted.)

### Server

Modbus-Server Node

### How can this be reproduced?

<details>
<summary>Example flow</summary>

```json
[
    {
        "id": "0e37c0f212591d47",
        "type": "modbus-read",
        "z": "d46d2bc29f86c250",
        "name": "Always running",
        "topic": "",
        "showStatusActivities": false,
        "logIOActivities": false,
        "showErrors": false,
        "showWarnings": true,
        "unitid": "",
        "dataType": "HoldingRegister",
        "adr": "0",
        "quantity": "1",
        "rate": "1",
        "rateUnit": "s",
        "delayOnStart": false,
        "startDelayTime": "",
        "server": "94d99994b72809c9",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "x": 490,
        "y": 380,
        "wires": [
            [
                "a89e59cbbad53d59"
            ],
            []
        ]
    },
    {
        "id": "a75206a1b33c393c",
        "type": "modbus-read",
        "z": "d46d2bc29f86c250",
        "name": "Right-click > Node > Disable me!",
        "topic": "",
        "showStatusActivities": false,
        "logIOActivities": false,
        "showErrors": false,
        "showWarnings": true,
        "unitid": "",
        "dataType": "HoldingRegister",
        "adr": "1",
        "quantity": "1",
        "rate": "1",
        "rateUnit": "s",
        "delayOnStart": false,
        "startDelayTime": "",
        "server": "94d99994b72809c9",
        "useIOFile": false,
        "ioFile": "",
        "useIOForPayload": false,
        "emptyMsgOnFail": false,
        "x": 520,
        "y": 500,
        "wires": [
            [
                "5dfd9114971bee9f"
            ],
            []
        ]
    },
    {
        "id": "24bff9ce523ba95d",
        "type": "comment",
        "z": "d46d2bc29f86c250",
        "name": "Right-click and disable the node below.",
        "info": "",
        "x": 540,
        "y": 440,
        "wires": []
    },
    {
        "id": "a89e59cbbad53d59",
        "type": "debug",
        "z": "d46d2bc29f86c250",
        "name": "debug 37",
        "active": true,
        "tosidebar": false,
        "console": false,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "payload",
        "statusType": "auto",
        "x": 680,
        "y": 380,
        "wires": []
    },
    {
        "id": "5dfd9114971bee9f",
        "type": "debug",
        "z": "d46d2bc29f86c250",
        "name": "debug 38",
        "active": true,
        "tosidebar": false,
        "console": false,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "payload",
        "statusType": "auto",
        "x": 760,
        "y": 500,
        "wires": []
    },
    {
        "id": "84a2aed2a22b5929",
        "type": "modbus-server",
        "z": "d46d2bc29f86c250",
        "name": "",
        "logEnabled": false,
        "hostname": "127.0.0.1",
        "serverPort": 10502,
        "responseDelay": 100,
        "delayUnit": "ms",
        "coilsBufferSize": 10000,
        "holdingBufferSize": 10000,
        "inputBufferSize": 10000,
        "discreteBufferSize": 10000,
        "showErrors": false,
        "showStatusActivities": false,
        "x": 480,
        "y": 240,
        "wires": [
            [],
            [],
            [],
            [],
            []
        ]
    },
    {
        "id": "66bc7cb59eb28303",
        "type": "comment",
        "z": "d46d2bc29f86c250",
        "name": "\"Deploy modified nodes\"",
        "info": "",
        "x": 500,
        "y": 560,
        "wires": []
    },
    {
        "id": "8e4ef4f6177cbd29",
        "type": "comment",
        "z": "d46d2bc29f86c250",
        "name": "Deploy this example",
        "info": "",
        "x": 490,
        "y": 320,
        "wires": []
    },
    {
        "id": "94d99994b72809c9",
        "type": "modbus-client",
        "name": "Test Server",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "queueLogEnabled": false,
        "failureLogEnabled": true,
        "tcpHost": "127.0.0.1",
        "tcpPort": "10502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": true,
        "showErrors": false,
        "showWarnings": true,
        "showLogs": true
    }
]
```

</details>

1. Import the flow above;
2. Deploy it;
3. Disable one of the `modbus-read` nodes (see the in-flow comments!)
4. Deploy the flow again.

### What did you expect to happen?

I would expect the remaining `modbus-read` nodes to continue operating and `modbus-server` to remain connected as long as there's at least one `modbus-read` node associated with the `modbus-server` is enabled.

grewek, 2024-11-07:
I took a look at the issue and i think i know what the issue is!

The Test Server client in the example is setup in both Nodes now one might think that these are separate clients but they are not! If you disable one of the nodes the client goes into it's closed state and that will affect the other node as well, i consider this a bug and we should change that behavior!

As a Workaround for now you could however create a new client for one of the nodes (the '+' symbol in the node settings) and set it to the same IP Address/Port this will decouple the clients!

grewek, 2024-11-15:
Good news!

I identified the bug and fixed it, but i still need to test things to see if everything else still works i expect a release in the next two weeks if that timetable changes i let you know.

Also thank you for the example flow and the very thorough explanation of the issue :)

Have a great weekend.

grewek, 2024-11-29:
Hi,
Bad news the release won't be coming this week :(
The issue is a bit more involved than i first thought so i need a bit more time!

My apologies for the short notice but i hoped that i can still find a fix today but that didn't worked out.

nagisa, 2024-11-29:
Hey,

Take your time! It really is something that only has affected me during development of flows, and most of my flows are quite stable at this point, so I don't see this come up very often.

Thanks for working on this!
S.

biancode, 2024-12-06:
option "d" for disable see "d": true => Disabled

closed 2025-03-07 by github-actions[bot]