BiancoRoyal/node-red-contrib-modbus #492: modbus-server's input message can not operate well,  So I modified the modbus-server-core.js , fixed these bugs

wei1224hf, 2024-09-07:
### Which node-red-contrib-modbus version are you using?

5.26.0

### What happened?

The Modbus-Server node , can not operate well if I input messages into this node , 
It operates well with modbus-read and modbus-write node , based on modbus-client node, 
but these operates are about AO and DO , can not operate DI and AI , 

If I want to modify the DI and AI data , I have to  use input discrete regisnter type to send message into the modbus-server node, 
although the DI and AI will change , but the address are wrong . 

So I modify the modbus-server-core.js , fixed the bugs , to make it work well:

'''
/**
 Copyright (c) since the year 2016 Klaus Landsdorf (http://plus4nodered.com/)
 All rights reserved.
 node-red-contrib-modbus

 @author <a href="mailto:klaus.landsdorf@bianco-royal.de">Klaus Landsdorf</a> (Bianco Royal)
 */
'use strict'
// SOURCE-MAP-REQUIRED

const _ = require('underscore')

// eslint-disable-next-line no-var
var de = de || { biancoroyal: { modbus: { core: { server: { } } } } } // eslint-disable-line no-use-before-define
de.biancoroyal.modbus.core.server.internalDebug = de.biancoroyal.modbus.core.server.internalDebug || require('debug')('contribModbus:core:server') // eslint-disable-line no-use-before-define

de.biancoroyal.modbus.core.server.bufferFactor = 2 //8
de.biancoroyal.modbus.core.server.memoryTypes = ['holding', 'coils', 'input', 'discrete']
de.biancoroyal.modbus.core.server.memoryUint16Types = ['holding', 'input']
de.biancoroyal.modbus.core.server.memoryUint8Types = ['coils', 'discrete']

de.biancoroyal.modbus.core.server.writeBit = function(buffer, byteIndex, bitPosition, value) {
    if (value) {
        // Set the bit (OR operation)
        buffer[byteIndex] |= (1 << bitPosition);
    } else {
        // Clear the bit (AND operation with the negated bit mask)
        buffer[byteIndex] &= ~(1 << bitPosition);
    }
}

de.biancoroyal.modbus.core.server.writeBitArray = function(currentBuffer, insertBuffer, position) {
    let byteIndex = Math.floor(position / 8); // Determine which byte to modify
    let bitPosition = position % 8; // Determine the starting bit position within the byte
    
    insertBuffer.forEach(bit => {
        if (bit === 1) {
            currentBuffer[byteIndex] |= (1 << (bitPosition)); // Set the bit at the position
        } else {
            currentBuffer[byteIndex] &= ~(1 << (bitPosition)); // Clear the bit
        }

        bitPosition++; // Move to the next bit

        if (bitPosition > 7) { // Move to the next byte if we've written 8 bits
            bitPosition = 0;
            byteIndex++;
        }
    });
    return currentBuffer;
}

de.biancoroyal.modbus.core.server.getLogFunction = function (node) {
  if (node.internalDebugLog) {
[... 174 lines cut ...]
    }
  } catch (err) {
    msg.error = err
    node.error(err)
  }
}

module.exports = de.biancoroyal.modbus.core.server
'''

### Server

Modbus-Server Node

### How can this be reproduced?

create a modbus-server node , send message into it with :
'''
msg.payload = { 'value': 1234, 'register': 'holding', 'address': 1 , 'disableMsgOutput' : 0 };
'''
then check the data with modbuscan32 ,  find there is a 1234 but the address is wrong.

### Other Information

I have already fixed the bugs by modifying the modbus-server-core.js

grewek, 2024-11-08:
hi @wei1224hf,

first of thank you for your contribution, do you think you could make a pull request? it's easier for me to take a look at the changes made :)

Thank you!

biancode, 2024-12-07:
@wei1224hf thank you for your contribution and for sharing your code! I've tested the changes and noticed some issues with the test suite. To maintain our quality standards and ensure compatibility for all users, we have a few requirements for contributions:

1. Please fork the repository and make your changes there
2. Ensure all tests pass (npm test)
3. Run the linter (npm run build)
4. Verify the changes don't break existing use cases
5. Submit a Pull Request with your improvements

I'm happy to review your PR and provide feedback. Let me know if you need any help with the testing setup or have questions about our contribution process.
Looking forward to your PR! 👍

see the [CONTRIBUTION.md](https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/master/.github/CONTRIBUTING.md)

biancode, 2024-12-07:
The changes are in the feature branch [core_server_changes](https://github.com/BiancoRoyal/node-red-contrib-modbus/tree/feature/core_server_changes)

closed 2025-03-09 by github-actions[bot]