BiancoRoyal/node-red-contrib-modbus #493: When trying to read several registers I get error: "fsm failed state after sending"

Kaiteki666, 2024-09-12:
### Which node-red-contrib-modbus version are you using?

5.30.0

### What happened?

Until a certain moment, node-red worked normally, without errors or complaints, reading a maximum of 27 registers at once, after some time of continuous operation, without touching node-red at all, the following problem arose:

When trying to read several registers at once, I get the error "fsm failed state after sending" and after that "Error: Timed out". Also less often, but it happens that an error occurs when trying to read 1 register at a time, the reasons are unknown.
I try to read the registers with Modbus Read and Modbus Sequencer nodes, their behavior is identical, it all depends on the number of registers.
When I read one register everything can be fine and after a few tries I can also get an error. 
After restarting node-red, when I try to read 10 registers at a time, the error does not occur for a while, but at some point it appears again and does not go away until I restart node-red. I can't read more than 10 registers at a time, I immediately get an error.
I tried changing the node-red version from older to newer, but that didn't help.

Here's the output I got when running node-red in debug mode with DEBUG=contribModbus*,modbus-serial node-red -v :

**1. Here is the output just after starting node-red:**
Welcome to Node-RED
===================

12 Sep 16:22:40 - [info] Node-RED version: v3.1.3
12 Sep 16:22:40 - [info] Node.js  version: v18.19.0
12 Sep 16:22:40 - [info] Linux 6.1.63-current-sunxi64 arm64 LE
12 Sep 16:22:42 - [info] Loading palette nodes
12 Sep 16:22:47 - [info] Dashboard version 3.6.2 started at /ui
12 Sep 16:22:48 - [info] Settings file  : /root/.node-red/settings.js
12 Sep 16:22:48 - [info] Context store  : 'memoryOnly' [module=memory]
12 Sep 16:22:48 - [info] Context store  : 'file' [module=localfilesystem]
12 Sep 16:22:49 - [info] User directory : /root/.node-red
12 Sep 16:22:49 - [warn] Projects disabled : editorTheme.projects.enabled=false
12 Sep 16:22:49 - [info] Flows file     : /root/.node-red/flows.json
12 Sep 16:22:49 - [info] Server now running at http://127.0.0.1:1880/
12 Sep 16:22:49 - [warn] Using unencrypted credentials
12 Sep 16:22:49 - [info] Starting flows
  contribModbus:queue:core queue serial lock command node name: Serial Switch USB0 to S0 id: f515fc0db2b361bf +0ms
  contribModbus:core:client Client -> new  +0ms
  contribModbus:core:client Client -> initialized  +1ms
12 Sep 16:22:49 - [warn] [modbus-read:545ad5d163805081] Read -> open node 545ad5d163805081 address: 0
  contribModbus:core:client Client -> new  +10ms
  contribModbus:core:client Client -> init  +2ms
12 Sep 16:22:49 - [warn] [modbus-client:Serial Switch USB0 to S0] Client -> fsm init state after new  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1
12 Sep 16:22:49 - [warn] [modbus-client:Serial Switch USB0 to S0] Client -> first fsm init in 500 ms  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +14ms
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +1ms
12 Sep 16:22:49 - [warn] [modbus-flex-sequencer:b69c1bbc1dbeb2b4] Flex-Sequencer -> initialize input delay timer node b69c1bbc1dbeb2b4
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +4ms
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +1ms
12 Sep 16:22:49 - [warn] [modbus-flex-sequencer:5627e60862f79b54] Flex-Sequencer -> initialize input delay timer node 5627e60862f79b54
12 Sep 16:22:49 - [warn] [modbus-read:80d0ec0fab64436e] Read -> open node 80d0ec0fab64436e address: 0
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +6ms
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +1ms
12 Sep 16:22:49 - [warn] [modbus-read:1d6b652aeb2ba950] Read -> open node 1d6b652aeb2ba950 address: 0
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +5ms
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +1ms
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +4ms
  contribModbus:core:client Client -> init  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +1ms
12 Sep 16:22:49 - [warn] [modbus-flex-sequencer:26d5497fbef3890c] Flex-Sequencer -> initialize input delay timer node 26d5497fbef3890c
12 Sep 16:22:49 - [info] Started flows
  contribModbus:core:client Client -> RTU buffered port serial  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +470ms
  contribModbus:core:client Client -> opened  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +19ms
  contribModbus:queue:core queue serial unlock command node name: Serial Switch USB0 to S0 id: f515fc0db2b361bf +544ms
  contribModbus:core:client Client -> time to open Unit 1  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +103ms
  modbus-serial modbus connection opened +0ms
  contribModbus:core:client Client -> connected  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1 +2ms
12 Sep 16:22:50 - [warn] [modbus-client:Serial Switch USB0 to S0] Client -> fsm connected after state opened Get More About It By Logging  Serial@/dev/ttyUSB0:115200bit/s default Unit-Id: 1
  contribModbus:queue:core queue serial unlock command node name: Serial Switch USB0 to S0 id: f515fc0db2b361bf +106ms
12 Sep 16:22:50 - [warn] [modbus-read:545ad5d163805081] Read -> startIntervalReading node 545ad5d163805081 address: 0
12 Sep 16:22:50 - [warn] [modbus-read:80d0ec0fab64436e] Read -> startIntervalReading node 80d0ec0fab64436e address: 0
12 Sep 16:22:50 - [warn] [modbus-read:1d6b652aeb2ba950] Read -> startIntervalReading node 1d6b652aeb2ba950 address: 0

[... 154 lines cut ...]
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB0",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "115200",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": "1",
        "commandDelay": "200",
        "clientTimeout": "1000",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": false,
        "showErrors": false,
        "showWarnings": true,
        "showLogs": true
    }
]

### What did you expect to happen?

I was expecting a successful read of multiple registers, with no errors

Kaiteki666, 2024-11-06:
Hello, the problem was solved by simply replacing the USB-RS485 converter

grewek, 2024-11-07:
Okay thank you for informing us about how you solved the issue :)

I close this issue for now and mark it as solved!

closed 2024-11-07 by grewek