BiancoRoyal/node-red-contrib-modbus #498: docker serial

zhushaoan, 2024-10-10:
### Which node-red-contrib-modbus version are you using?

4.0.3 docker arm

### What happened?

```
version: '3.8'

services:
  nodered:
    image: registry.xiangyuniot.com/edge/nodered:4.0.3-arm
    container_name: nodered
    user: root
    network_mode: host
    cap_add:
      - NET_ADMIN
    volumes:
      - node-red-data:/data
      - /sys/class/leds:/sys/class/leds
      - /sys/devices/platform/leds:/sys/devices/platform/leds
      - /dev/ttyS3:/dev/ttyS3_4851
      - /dev/ttyS4:/dev/ttyS4_485422
      - /dev/ttyS9:/dev/ttyS9_232
    tty: true
    stdin_open: true
    
volumes:
  node-red-data:
```

```
 docker logs f99e0dd2bad3
10 Oct 07:11:40 - [info]

Welcome to Node-RED
===================

10 Oct 07:11:40 - [info] Node-RED version: v4.0.3
10 Oct 07:11:40 - [info] Node.js  version: v20.17.0
10 Oct 07:11:40 - [info] Linux 5.10.110-rockchip-rk3588 arm64 LE
10 Oct 07:11:40 - [info] Loading palette nodes
10 Oct 07:11:41 - [info] Settings file  : /data/settings.js
10 Oct 07:11:41 - [info] Context store  : 'default' [module=memory]
10 Oct 07:11:41 - [info] User directory : /data
10 Oct 07:11:41 - [warn] Projects disabled : editorTheme.projects.enabled=false
10 Oct 07:11:41 - [info] Flows file     : /data/flows.json
10 Oct 07:11:41 - [warn]
```

### Server

None/This is related to a node that doesn't connect to a server

### How can this be reproduced?

-----------------------------------

```
root@hanweiai200:/home/hanweiai# ls /dev/ttyS* /dev/ttyUSB*
/dev/ttyS0  /dev/ttyS3  /dev/ttyS4  /dev/ttyS9  /dev/ttyUSB0  /dev/ttyUSB1  /dev/ttyUSB2  /dev/ttyUSB3
root@hanweiai200:/home/hanweiai# docker exec -it f99e0dd2bad3 /bin/bash
hanweiai200:/usr/src/node-red# ls /dev/ttyS* /dev/ttyUSB*
ls: /dev/ttyUSB*: No such file or directory
/dev/ttyS3_4851    /dev/ttyS4_485422  /dev/ttyS9_232
hanweiai200:/usr/src/node-red# ls -l /dev/ttyS3
ls: /dev/ttyS3: No such file or directory
hanweiai200:/usr/src/node-red# ls -l /dev/ttyS3_4851
crw-rw----    1 root     dialout     4,  67 Oct 10 08:28 /dev/ttyS3_4851

```
[image]
Why is the serial port list displayed the serial port list of the host machine instead of the docker container?

zhushaoan, 2024-10-15:
try another noede "node-red-node-serialport"， it can work ok!
But “node-red-contrib-modbus”：
./entrypoint.sh: line 14:     7 Segmentation fault      (core dumped) /usr/local/bin/node $NODE_OPTIONS node_modules/node-red/red.js --userDir /data $FLOWS "${@}"

```

    {
        "id": "eed8a0b1c2bc58f1",
        "type": "serial-port",
        "name": "",
        "serialport": "/dev/ttyS3",
        "serialbaud": "4800",
        "databits": "8",
        "parity": "none",
        "stopbits": "1",
        "waitfor": "",
        "dtr": "none",
        "rts": "none",
        "cts": "none",
        "dsr": "none",
        "newline": "\\r\\n",
        "bin": "false",
        "out": "char",
        "addchar": "",
        "responsetimeout": "10000"
    },
    {
        "id": "19ec058c2ac710c8",
        "type": "modbus-client",
        "name": "",
        "clienttype": "serial",
        "bufferCommands": true,
        "stateLogEnabled": true,
        "queueLogEnabled": true,
        "failureLogEnabled": true,
        "tcpHost": "127.0.0.1",
        "tcpPort": "502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyS3",
        "serialType": "RTU",
        "serialBaudrate": "4800",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "serialAsciiResponseStartDelimiter": "0x3A",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectOnTimeout": true,
        "reconnectTimeout": "2000",
        "parallelUnitIdsAllowed": true,
        "showErrors": true,
        "showWarnings": true,
        "showLogs": true
    },
```

schefa11, 2025-01-28:
I have the same issue with a Node-RED 4.0.3 and node-red-contrib-modbus version 5.42.0.

Installing the node via `npm install --omit=dev --unsafe-perm --build-from-source` solves that issue for me.

If the node is already installed you can probably run `npm rebuild`

closed 2025-05-14 by github-actions[bot]