BiancoRoyal/node-red-contrib-modbus #502: Flex Writer "Error: Modbus exception 3: Illegal data value (value cannot be written to this register)"

OverWorldD, 2024-11-16:
### Which node-red-contrib-modbus version are you using?

5.14.1

### What happened?

I'm having problems with the flex writer not working, the modbus response shows "" and a debug node says "Error: Modbus exception 3: Illegal data value (value cannot be written to this register)"

this is the object I'm feeding it:

{"value":[5],"fc":16,"unitid":2,"address":48444,"quantity":1}

I got it to work once, It seems intermittent. I cannot understand what is causing it. The flex write node works on other flows.
I'm 

### Server

None/This is related to a node that doesn't connect to a server

### How can this be reproduced?

[error.json](https://github.com/user-attachments/files/17782690/error.json)

### Other Information

NodeJS v14.21.3
NodeRed 1.3.7

OverWorldD, 2024-11-27:
Just a small update here that the error persists and I have still not been able to find a solution

grewek, 2024-11-29:
Hey i take a look at it next week! You could also try to activate the error output of the node to see what it says maybe the device does not allow writing to that particular register?

grewek, 2024-12-05:
I just checked the described behavior with a Simulated Modbus Server, it worked without issues! My best guess is that the connection to the device is unstable? That would explain why the  problem is intermittent.

Also my apologies for my last answer i was in a rush and didn't see that you already posted the thrown error message by the client.

OverWorldD, 2024-12-05:
Thank you! I tried to re-build the flow from scratch. It throws the error now, but the address is actually written. It might be that I'm overloading the bus I have pretty much communication going on. I also upgraded node-red and node-js, running on an old raspbberry pi so I'm a bit limited to what versions i can run, still on node-red 1.3.7 and node.js 14. This started after the upgrade, i had some dependency issues. Could it be affected by the version of serialport?

grewek, 2024-12-17:
Hi,
What you can try is to  send messages slow(ish) (> 100ms interval) so that the client queue isn't overburdened! Also check our [Versions.md](https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/master/HISTORY.md) to see if you could run into any compatibility issues! If i should guess i would definitely look at the speed that the messages are send at.

closed 2025-03-18 by github-actions[bot]