BiancoRoyal/node-red-contrib-modbus #509: Modbus Flex FC shows "broken" without map defined

unclej84, 2024-12-19:
### Which node-red-contrib-modbus version are you using?

5.43.0

### What happened?

When creating a new Flex FC node with only server and unitId defined but maps just filled with `{}` (empty) for use with maps defined on incoming messages, all mobdus nodes get status `broken` shortly after deploy. Maybe this is related to #508 as it tries to process empty request? This can only be fixed by having some map defined in the node even when not needed.

### Server

Other/External server

### How can this be reproduced?

create a new node, enter server and unitId without defining any map (default `{}`) and deploy

### What did you expect to happen?

Default map without configuration should not break the modbus nodes / communication

grewek, 2024-12-20:
Yes that seems like a bug, will put this on the list for things todo.
Thank you for the report!
This is also probably related to another bug i am currently working (our internal state-machine).
I will check this after Vacation.

Happy Holidays :)

unclej84, 2025-03-12:
just to remove stale label

closed 2025-08-11 by github-actions[bot]