BiancoRoyal/node-red-contrib-modbus #510: Sequencer msg.topic = name

rmv5004, 2024-12-23:
### Describe the requested feature

The sequencer msg.topic should be the name of the sequence that was processed.  Currently you need to parse the full message.  If msg.topic = name it would simplify the process where you can use the msg.payload.

[image]

### Motivation

Ease of use

biancode, 2025-01-10:
This should come from the inject node topic or from a function node after. I'm sorry ,but your idea is not Node-RED conform.

rmv5004, 2025-01-16:
That makes the sequencer payload worthless you may as well use separate read / writes.  What about an option to change the topic for each sequence?  I have seen several nodes with that option.

[image]

closed 2025-04-16 by github-actions[bot]