BiancoRoyal/node-red-contrib-modbus #525: When the read cycle is made faster in Modbus TCP, writing is no longer performed

shamada-idec, 2025-04-07:
While I was periodically reading 200 words of data using Modbus Read, I noticed that Modbus Write was not being executed randomly.
When the Modbus Read cycle is set to 400 ms, it works without any problems, but when the cycle is set to 200 ms, Modbus Write may or may not be executed.

Can you tell me some solution to this issue?

biancode, 2025-05-21:
Many devices have some restriction in communication speed and data amount to process per cycle. If you can open two connections try to use one to read and the other to write and keep in mind to get no dependency in those operations.

shamada-idec, 2025-05-21:
Thank you for reply. I will try to change connections between writing and reading.

closed 2025-05-21 by shamada-idec