BiancoRoyal/node-red-contrib-modbus #540: Modbus-Read node not working after update Node-Red to 4.1.0 version

ftaddia, 2025-08-17:
### Which node-red-contrib-modbus version are you using?

5.43.0

### What happened?

After updating to Node-Red 4.1.0, reading a variable via modbus no longer returns the correct value but 0 or values like 32000.

The variable is read via Modbus using the "FC 3: Read Holding Registers" type. I should always receive a 3-digit number, but instead I receive either 0 or very high numbers like 32000.

**I'd like to point out that only reading variables with "FC 3: Read Holding Registers" isn't working.**

This is my configuration and on the right the logs of the values I said were wrong:

[image]

### Server

Modbus-Server Node

### How can this be reproduced?

I created a special flow to export the json so you can replicate my exact flow.

[issue_flow.json](https://github.com/user-attachments/files/21823329/issue_flow.json)

### What did you expect to happen?

What I expect is that if in the original variable there is a string or a number for example "308", reading that variable with modbus-read should show me the same value.

### Other Information

Node-Red version: 4.1.0
NodeJS version: v22.18.0

I've already tried reinstalling NodeJS 20, but the result is the same.
I didn't downgrade my Node-Red version, but the problem started right after I upgraded.

colinl, 2025-08-17:
It would be worth downgrading node-red to check it is not just a coincidence.  Assuming that you installed globally and are running on a linux system then that is as simple as
`sudo npm install --unsafe-perm node-red@4.0.9`
and restart node-red.

ftaddia, 2025-08-17:
Yes, I'm on a Debian system.
I usually run updates with this: bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered)
Also, in the user's home directory, there's a .node-red folder with all the configuration and flows.
I tried running the command sudo npm install --unsafe-perm node-red@4.0.9, but after rebooting, it's still at version 4.1.0. I also don't know if running this command corrupted my current node-red installation.

ftaddia, 2025-08-19:
@grewek I'm checking the Modbus connectors leading to the PLC, and I'll also use a tester to verify that they're getting power, making sure the cables are working properly. I'll let you know. Thank you all very much.

ftaddia, 2025-08-20:
I rewired the cables used for communication, but the error persists @grewek.
I saw that a commit has been linked to this issue. When do you plan to release it to production in a new version @biancode ?

grewek, 2025-08-28:
Sorry for the late reply, i checked our tests and debugged it, the issue is happening there as well i don't know why the test system says that it is passing but i will debug it and fix it asap.

grewek, 2025-09-03:
@ftaddia 

Can you check with another tool that can do modbus reads and see if the results are the same? I really need more information in order to find the actual issue!

commit ac86eef6 (2025-08-20): fix: Node-RED 4.1.0 buffer compatibility and memory leak on Exception 11

- Add cloneBuffer utility to prevent buffer corruption in Node-RED 4.1.0+
- Clone response buffers before passing them in messages
- Add automatic queue cleanup on Exception 11 (Gateway Target Device Failed to Respond)
- Prevent memory accumulation when devices fail to respond

Fixes #540, #536