BiancoRoyal/node-red-contrib-modbus #541: "Error: Data length error, expected 7 got 9"

grasgruenit, 2025-09-02:
Hi,

I am using Modbus TCP to read data from a district heating system interface (Schneid MR-12). I am able to read many registers without an issue. When trying to read the meter data from register 730 I get following error:
Error: Data length error, expected 7 got 9

When reading quantity 2, the error is:
Error: Data length error, expected 9 got 13

Regardless if I use FC3 or FC4. Also no difference if I change the TCP Type from DEFAULT to RTU-buffered. Other types are failing completely.

grasgruenit, 2025-09-03:
Hey @grewek,

thank you for the answer.
I am not shure how the data is transfered internally in the device. I am accessing it with TCP.

Did now install Wireshark on my computer and used ModbusTcpTool to grab data from the register. Do you see anything?

282	8.838238	192.168.0.55	192.168.0.199	TCP	66	18603 → 2001 [SYN] Seq=0 Win=65535 Len=0 MSS=1460 WS=256 SACK_PERM
283	8.839417	192.168.0.199	192.168.0.55	TCP	60	2001 → 18603 [SYN, ACK] Seq=0 Ack=1 Win=5744 Len=0 MSS=1440
284	8.839511	192.168.0.55	192.168.0.199	TCP	54	18603 → 2001 [ACK] Seq=1 Ack=1 Win=65535 Len=0
285	8.842794	192.168.0.55	192.168.0.199	TCP	66	18603 → 2001 [PSH, ACK] Seq=1 Ack=1 Win=65535 Len=12
286	8.856072	192.168.0.199	192.168.0.55	TCP	71	2001 → 18603 [PSH, ACK] Seq=1 Ack=13 Win=5732 Len=17
288	8.901056	192.168.0.55	192.168.0.199	TCP	54	18603 → 2001 [ACK] Seq=13 Ack=18 Win=65518 Len=0
512	11.325907	192.168.0.55	192.168.0.199	TCP	66	18603 → 2001 [PSH, ACK] Seq=13 Ack=18 Win=65518 Len=12
513	11.330460	192.168.0.199	192.168.0.55	TCP	71	2001 → 18603 [PSH, ACK] Seq=18 Ack=25 Win=5720 Len=17
514	11.385537	192.168.0.55	192.168.0.199	TCP	54	18603 → 2001 [ACK] Seq=25 Ack=35 Win=65501 Len=0
1704	15.331721	192.168.0.199	192.168.0.55	TCP	60	2001 → 18603 [FIN, ACK] Seq=35 Ack=25 Win=5720 Len=0
1705	15.331797	192.168.0.55	192.168.0.199	TCP	54	18603 → 2001 [ACK] Seq=25 Ack=36 Win=65501 Len=0

grewek, 2025-09-03:
[image]

grewek, 2025-09-04:
Ok the issue is that there are four trailing bytes at the end of the modbus response, i don't know why they are there and what it's purpose are. But this issue should be fixable we will just add a check that if the byte count is correct (FC1,FC2,FC3 and FC4) we parse the message anyway.

Thank you for the bug report and support in getting the necessary data to debug this.

grasgruenit, 2025-09-14:
> The new version is now released, we had to gate this feature because it
> could potentially break stuff for other user so in order to activate it you
> need to open the properties of the read node in question and activate
> 'allow oversized message support'.
>
> *NOTE*: We only support FC1, FC2, FC3 and FC4 because these function
> codes are the only one that carry an internal byte count. Also you find
> this setting only in the Nodes modbus-read, modbus-getter and
> modbus-flex-getter
>
> Please let me know if it works for you or if you still have issues

Hi Kay,

thank you. It works. I can read the content of the fields.

Any chance to disable the warning each time It reads a value?
"Deformed Message support is enabled"

This is shown altough I did disable errors and warnings.

Best regards

/ Manuel Glojek | +436643256932
Zeit und Platz sparende Kurzsignatur. Ich bitte um Nachsicht für eventuell
fehlende persönliche Anrede und Grüße.

grewek, 2025-09-16:
Yeah these messages should be gone when you click the disable warnings button. I think i messed something up will investigate and push a fix.

grasgruenit, 2025-09-17:
Perfect work. Thank you.